SpatialVLA/SpatialVLA #1: Weight Decay Question

mbreuss, 2025-02-04:
Hi,

Congrats for the amazing paper and thanks a lot for sharing the full codebase!
I got two general questions regarding your hyperparameters during pretrainig: Did you ablate the weight decay of 0 against trainings with higher weight decay? Or can you tell your intuition about this design decision? 
Did you also test trainings with a lower batch size or have any experience how that influences cross-embodiment pretraining success?

Thanks a lot!

DelinQu, 2025-02-05:
Hi [mbreuss](https://github.com/mbreuss),

Thanks for your attention. We didn't perform ablation studies on the weight decay of 0 specifically. 
Initially, we followed the pretraining parameters of multimodal models like LLaMA and [LLaVA](https://github.com/haotian-liu/LLaVA/blob/main/scripts/pretrain.sh), including learning rate, warmup ratio, and cosine scheduler. However, we observed that this led to unstable training. To address this, we incorporated training techniques from [OpenVLA](https://github.com/openvla/openvla/blob/main/vla-scripts/train.py), including a weight decay of 0 and a learning rate of 2e-5. 

In my experience, weight decay is not a crucial factor affecting training performance. Instead, the dataset, module freezing, and learning rate tend to have a more significant impact. Nevertheless, I'm open to exploring the effect of weight decay in our specific setup. I'd be happy to train on the Bridge and Fractal mixed dataset, which may take a few days :).

mbreuss, 2025-02-05:
Thanks for sharing your experiences! :) 

> we followed the pretraining parameters of multimodal models like LLaMA and [LLaVA](https://github.com/haotian-liu/LLaVA/blob/main/scripts/pretrain.sh), including learning rate, warmup ratio, and cosine scheduler.

Very interesting,  I expected consine scheduler and warmup to help.

> module freezing
Are you referring to the LLM embeddings here? Did you do some ablations for the performance differences?

> Nevertheless, I'm open to exploring the effect of weight decay in our specific setup.

Thanks, but please don't sacrifice so many GPU hours because of this question :D

DelinQu, 2025-02-05:
> module freezing
> Are you referring to the LLM embeddings here? Did you do some ablations for the performance differences?

In fact, full-parameter training yields significantly better results during deployment. If we freeze vision modules, such as SigLip, the model may even fail to converge and become extremely unstable in deployment, e.g., weird grasping trajectory and fail to recognize objects. While freezing the LLM embeddings does provide some improvement, it's not the primary factor contributing to the model's success.

mbreuss, 2025-02-05:
Thanks a lot for sharing! 
One final question regarding the training duration. Did the model accuracy stagnate in the end or how did you decide to stop training?

DelinQu, 2025-02-05:
> Thanks a lot for sharing! One final question regarding the training duration. Did the model accuracy stagnate in the end or how did you decide to stop training?

Duration training, we set a standard of training for 1 epoch, at which point the accuracy reached around 95%. We found that the 60% progress checkpoint already performed well on both simpler env benchmarks and real experiments, with no extramly difference from the 100% progress checkpoint. 

The question of when to terminate training is indeed debatable. In both OpenVLA and SpatialVLA, we stopped training at 95% accuracy (1 epoch), but I'm also aware that Google's RT series training experience suggests that longer training is often better, even if accuracy plateaus or decreases. For large-scale datasets like OpenX, I believe that 1-2 epoch training is a safe and reasonable choice.

closed 2025-02-05 by DelinQu

mbreuss, 2025-02-05:
Thank you for the detailed answers and congrats to the awesome project again!:)