SpatialVLA/SpatialVLA #3: RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent

cheng-haha, 2025-02-06:
I am conducting evaluations on a server without a display, but I keep encountering the error mentioned above. Currently, I have tried creating a virtual display based on [https://github.com/simpler-env/SimplerEnv/issues/18](18), but it doesn't seem to work. **Have you encountered this issue before, and how can it be resolved?**

Full error message:

(simpler_env) [root:SimplerEnv-OpenVLA]$ bash scripts/run_spatialvla.sh
🚀 running bridge.sh ...
2025-02-07 02:50:32.635028: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-02-07 02:50:32.676287: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-02-07 02:50:32.676340: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-02-07 02:50:32.677505: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-02-07 02:50:32.684244: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
2025-02-07 02:50:33.397145: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:38] TF-TRT Warning: Could not find TensorRT
Octo is not correctly imported.
No module named 'octo'
2025-02-07 02:50:35.765549: W tensorflow/core/common_runtime/gpu/gpu_device.cc:2256] Cannot dlopen some GPU libraries. Please make sure the missing libraries mentioned above are installed properly if you would like to use GPU. Follow the guide at https://www.tensorflow.org/install/gpu for how to download and setup the required libraries for your platform.
Skipping registering GPU devices...
**** spatialvla ****
*** policy_setup: widowx_bridge, unnorm_key: bridge_orig/1.0.0 ***
Some kwargs in processor config are unused and will not have any effect: num_obs_steps, intrinsic_config, bin_policy, obs_delta, statistics, action_config, action_chunk_size. 
scale intrinsic of bridge_orig/1.0.0 from [[623.588, 0, 319.501], [0, 623.588, 239.545], [0, 0, 1]] to tensor([[218.2558,   0.0000, 111.8253],
        [  0.0000, 291.0078, 111.7877],
        [  0.0000,   0.0000,   1.0000]]) ...
scale intrinsic of default from [[623.588, 0, 319.501], [0, 623.588, 239.545], [0, 0, 1]] to tensor([[218.2558,   0.0000, 111.8253],
        [  0.0000, 291.0078, 111.7877],
        [  0.0000,   0.0000,   1.0000]]) ...
Add 0 TRANSLATION TOKENS, tokenizer vocab size 257152 / 265347
Add 0 ROTATION TOKENS to tokenizer, tokenizer vocab size 257152 / 265347
Add 0 GRIPPER TOKENS to tokenizer, tokenizer vocab size 257152 / 265347
🔥 init from default configurations ... None
Loading checkpoint shards: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2/2 [00:00<00:00,  2.60it/s]
Control mode:  arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos
[2025-02-07 02:50:50.967] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2025-02-07 02:50:50.967] [svulkan2] [warning] Continue without GLFW.
Traceback (most recent call last):
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/simpler_env/main_inference.py", line 85, in <module>
    success_arr = maniskill2_evaluator(model, args)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/simpler_env/evaluation/maniskill2_evaluator.py", line 227, in maniskill2_evaluator
    run_maniskill2_eval_single_episode(
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/simpler_env/evaluation/maniskill2_evaluator.py", line 62, in run_maniskill2_eval_single_episode
    env = build_maniskill2_env(
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/simpler_env/utils/env/env_builder.py", line 16, in build_maniskill2_env
    env = gym.make(env_name, **kwargs)
  File "/opt/conda/envs/simpler_env/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 740, in make
    env = env_creator(**env_spec_kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 92, in make
    env = env_spec.make(**kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/utils/registration.py", line 34, in make
    return self.cls(**_kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 310, in __init__
    super().__init__(
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/put_on_in_scene.py", line 162, in __init__
    super().__init__(**kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/move_near_in_scene.py", line 58, in __init__
    super().__init__(**kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/envs/custom_scenes/base_env.py", line 134, in __init__
    super().__init__(**kwargs)
  File "/inspire/hdd/ws-f4d69b29-e0a5-44e6-bd92-acf4de9990f0/public-project/chengdongzhou-240108390137/vla_projects/SimplerEnv-OpenVLA/ManiSkill2_real2sim/mani_skill2_real2sim/envs/sapien_env.py", line 107, in __init__
    self._renderer = sapien.SapienRenderer(**renderer_kwargs)
RuntimeError: vk::PhysicalDevice::createDeviceUnique: ErrorExtensionNotPresent
2025-02-07 02:50:52.858202: I tensorflow/core/util/port.cc:113] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-02-07 02:50:52.900385: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
2025-02-07 02:50:52.900434: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:607] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
2025-02-07 02:50:52.901635: E external/local_xla/xla/stream_executor/cuda/cuda_blas.cc:1515] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-02-07 02:50:52.908447: I tensorflow/core/platform/cpu_feature_guard.cc:182] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX512F AVX512_VNNI AVX512_BF16 AVX_VNNI AMX_TILE AMX_INT8 AMX_BF16 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.

DelinQu, 2025-02-07:
Hi, 
You can try checking the following aspects:

1. Check if `vulkaninfo` is working correctly. The troubleshooting in 
 https://maniskill.readthedocs.io/en/latest/user_guide/getting_started/installation.html#vulkan Install the libvulkan1 and set the Nvidia configuration files. 
```bash
XDG_RUNTIME_DIR=/usr/lib
LD_LIBRARY_PATH="/usr/lib/x86_64-linux-gnu:${LD_LIBRARY_PATH}"
vulkaninfo # check
```
If you encounter issues, please also check my customized deployed scrips, which install everything in an A100 cluster.

[vulkan_setup.zip](https://github.com/user-attachments/files/18699962/vulkan_setup.zip)

2. Check the package version of the gym and sapien package. In simpler env evaluation, our environments use SAPIEN 2.2 and ManiSkill2, not SAPIEN 3 and ManiSkill3. It also leads to environment-creating failures if you get the incorrect version.
```bash
gym                          0.23.0
sapien                       2.2.2
```
Share the logging and I will try my best to solve the issues 😄

closed 2025-02-23 by DelinQu

Luo-Zhongwei, 2025-03-29:
hi,how you solve it, i meet the same problem