SpatialVLA/SpatialVLA #5: Could you release the LIBERO evaluation code?

QuanyiLi, 2025-02-19:
Hi SpatialVLA team,

I am looking for a SOTA VLA to play with and came across SpatialVLA just now. I believe this is the best one I can find!

I find that the SimplerEnv evaluation code has been provided. Is there any plan to release the LIBERO evaluation code as well?

DelinQu, 2025-02-19:
yes, but the current sota on libero is pi0, much better than current methods 😄. I think the performance comes from the integration of state information. I spend some time playing with three versions of pi0: 
- https://github.com/Physical-Intelligence/openpi
- https://github.com/allenzren/open-pi-zero
- https://github.com/huggingface/lerobot/tree/main/lerobot/common/policies/pi0
yes, amazing. The model performs well, achieving approximately 94% accuracy on average, which is almost identical to the official evaluation. So, if you're looking for policies that perform best in libero, pi0 would be the best choice. The next version of spatialval will integrate with robot state modality and will finish the pre-training in a few months. I got to know in an email that the huggingface lerobot team already has an implementation for LIBERO in LeRobot and will be released soon.

The following links would also be helpful:
- https://x.com/delin_qu23107/status/1891332054572318874
- https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/README.md

[image]

QuanyiLi, 2025-02-19:
Thank you for the reply. I am looking at \pi0 just now after finding your tweet : ) small world

closed 2025-02-19 by QuanyiLi