SpatialVLA/SpatialVLA #6: How to use the rh20t data, and is this part open source?

KeyuLi, 2025-02-19:


DelinQu, 2025-02-19:
Yes, we converted RH20T into RLDS ourselves. We can provide the script or the complete converted dataset, which is about 3.9TB. Additionally, our recent project https://huggingface.co/spaces/IPEC-COMMUNITY/openx_dataset_lerobot_v2.0 is converting RH20T into Lerobot’s format to facilitate large-scale pretraining for the community. This may take a few days to a week

KeyuLi, 2025-02-20:
Thank you for your reply and your contribution to the open source community. Additionally, I would like to ask if https://huggingface.co/spaces/IPEC-COMMUNITY/openx_dataset_lerobot_v2.0 provides data converted from OXE and RH20T to LeRobot format? I don't seem to see the RH20T data - is this still in progress? Also, would it be possible to provide the scripts for converting OXE and RH20T to LeRobot format? I would be very grateful if you could share these (as I have already downloaded the original versions of OXE and RH20T).

DelinQu, 2025-02-20:
The pretraining data conversion (rlds/others -> Lerobot) of OXE is almost done, and the RH20T is still in progress :). You can refer to https://github.com/Tavish9/openx2lerobot to get the conversion scripts. And, we'll release the code related to RH20T as soon as possible :)

Tavish9, 2025-02-20:
Hi @KeyuLi, 

Just a quick reminder: [lerobot](https://github.com/huggingface/lerobot) is actively updating LeRobotDataset, so the script might be incompatible with the new version of LeRobot. We will stay up to date with the official updates and adjust the script as necessary. Please make sure to follow the latest version of the [script](https://github.com/Tavish9/openx2lerobot).

KeyuLi, 2025-02-20:
I understand completely. Data format standardization is truly tedious but essential work. Thank you for your efforts on this. I'll be following your work at https://huggingface.co/IPEC-COMMUNITY and looking forward to future updates!

closed 2025-02-20 by KeyuLi