SpatialVLA/SpatialVLA #7: RuntimeError: Error building extension 'fused_adam'

githubLeoliu, 2025-02-23:
I am trying to go through the pretrain step by running scripts/spatialvla_4b_pretrain/torchrun_pretrain.sh. I am getting the following error: 

```
🔥 init from default configurations ... None
Using /home/leoliu/.cache/torch_extensions/py310_cu121 as PyTorch extensions root...
ERROR: ld.so: object '../libtcmalloc.so.4.5.3' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ERROR: ld.so: object '../libtcmalloc.so.4.5.3' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Detected CUDA files, patching ldflags
Emitting ninja build file /home/leoliu/.cache/torch_extensions/py310_cu121/fused_adam/build.ninja...
Building extension module fused_adam...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ERROR: ld.so: object '../libtcmalloc.so.4.5.3' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
ninja: error: unknown target '–v'
🔥 init from default configurations ... None
[rank0]: Traceback (most recent call last):
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 2104, in _run_ninja_build
[rank0]:     subprocess.run(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/subprocess.py", line 526, in run
[rank0]:     raise CalledProcessError(retcode, process.args,
[rank0]: subprocess.CalledProcessError: Command '['ninja', '–v']' returned non-zero exit status 1.

[rank0]: The above exception was the direct cause of the following exception:

[rank0]: Traceback (most recent call last):
[rank0]:   File "/home/leoliu/VLA/SpatialVLA/train/spatialvla_pretrain.py", line 697, in <module>
[rank0]:     main()
[rank0]:   File "/home/leoliu/VLA/SpatialVLA/train/spatialvla_pretrain.py", line 665, in main
[rank0]:     train_result = trainer.train(resume_from_checkpoint=checkpoint)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/transformers/trainer.py", line 2164, in train
[rank0]:     return inner_training_loop(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/transformers/trainer.py", line 2325, in _inner_training_loop
[rank0]:     model, self.optimizer, self.lr_scheduler = self.accelerator.prepare(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/accelerate/accelerator.py", line 1344, in prepare
[rank0]:     result = self._prepare_deepspeed(*args)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/accelerate/accelerator.py", line 1851, in _prepare_deepspeed
[rank0]:     engine, optimizer, _, lr_scheduler = ds_initialize(**kwargs)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/__init__.py", line 193, in initialize
[rank0]:     engine = DeepSpeedEngine(args=args,
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/runtime/engine.py", line 313, in __init__
[rank0]:     self._configure_optimizer(optimizer, model_parameters)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/runtime/engine.py", line 1276, in _configure_optimizer
[rank0]:     basic_optimizer = self._configure_basic_optimizer(model_parameters)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/runtime/engine.py", line 1353, in _configure_basic_optimizer
[rank0]:     optimizer = FusedAdam(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/ops/adam/fused_adam.py", line 94, in __init__
[rank0]:     fused_adam_cuda = FusedAdamBuilder().load()
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/ops/op_builder/builder.py", line 531, in load
[rank0]:     return self.jit_load(verbose)
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/deepspeed/ops/op_builder/builder.py", line 578, in jit_load
[rank0]:     op_module = load(name=self.name,
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 1314, in load
[rank0]:     return _jit_compile(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 1721, in _jit_compile
[rank0]:     _write_ninja_file_and_build_library(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 1833, in _write_ninja_file_and_build_library
[rank0]:     _run_ninja_build(
[rank0]:   File "/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 2120, in _run_ninja_build
[rank0]:     raise RuntimeError(message) from e
[rank0]: RuntimeError: Error building extension 'fused_adam'
```

The related package version are following:

```
transformers              4.47.0                   pypi_0    pypi
torch                     2.5.1+cu121              pypi_0    pypi
torchaudio                2.5.1+cu121              pypi_0    pypi
torchvision               0.20.1+cu121             pypi_0    pypi
deepspeed                 0.15.3                   pypi_0    pypi
deepspeed-kernels         0.0.1.dev1698255861          pypi_0    pypi

gcc --version
gcc (Ubuntu 7.5.0-6ubuntu2) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

nvidia-driver version:
Driver Version: 555.42.06
```

Does anyone know how to fix this issue?

DelinQu, 2025-02-23:
Hi [githubLeoliu](https://github.com/githubLeoliu), this relates to the environment variables in the DeepSpeed fuse Adam building. Could you please share your PATH, LD_LIBRARY_PATH, CUDA_HOME, and the version of Ninja?  Additionally, below are my extra environment variables setting for your reference:

```bash
# for torch>=2.5.0 in my case
export PATH="/mnt/vla_ptm/miniconda3/bin:$PATH"
export LD_LIBRARY_PATH="/mnt/vla_ptm/miniconda3/lib:${LD_LIBRARY_PATH}"
export LD_LIBRARY_PATH="/mnt/vla_ptm/miniconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/nvjitlink/lib:${LD_LIBRARY_PATH}"
export LD_LIBRARY_PATH="/mnt/vla_ptm/miniconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/cusparse/lib:${LD_LIBRARY_PATH}"

CFLAGS="/cpfs01/shared/optimal/vla_ptm/miniconda3/include"
LDFLAGS="/cpfs01/shared/optimal/vla_ptm/miniconda3/lib"
```
- CUDA_HOME: /usr/local/cuda/
- NVCC version: 12.4
- LD_LIBRARY_PATH:
```
/usr/lib/x86_64-linux-gnu:/mnt/vla_ptm/miniconda3/envs/vla_next/lib/python3.10/site-packages/nvidia/cusparse/lib:/mnt/vla_ptm/miniconda3/envs/vla_next/lib/python3.10/site-packages/nvidia/nvjitlink/lib:/mnt/vla_ptm/miniconda3/lib:/usr/local/nvidia/lib:/usr/local/nvidia/lib64
```
- PATH:
```
/mnt/vla_ptm/miniconda3/envs/vla_next/bin:/mnt/vla_ptm/miniconda3/bin:/mnt/vla_ptm/miniconda3/condabin:/usr/local/bin:/usr/local/bin:/usr/local/nvidia/bin:/usr/local/cuda/bin
```
It can be crucial to ensure that `PATH/CUDA_HOME` and `LD_LIBRARY_PATH` are consistent, according to [
Error building extension 'fused_adam' with DeepSpeed==0.3.13](https://github.com/deepspeedai/DeepSpeed/issues/885#issuecomment-810748368).

githubLeoliu, 2025-02-23:
After adding the following lines into my .bashrc file

```
# SpatialVLA ENV
export PATH="/home/leoliu/anaconda3/bin:$PATH"
export LD_LIBRARY_PATH="/home/leoliu/anaconda3/lib:${LD_LIBRARY_PATH}"
export LD_LIBRARY_PATH="/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/nvjitlink/lib:${LD_LIBRARY_PATH}"
export LD_LIBRARY_PATH="/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/cusparse/lib:${LD_LIBRARY_PATH}"

CFLAGS="/home/leoliu/anaconda3/include"
LDFLAGS="/home/leoliu/anaconda3/lib"
```

I have:
```
echo $PATH
/home/leoliu/anaconda3/envs/spatialvla/bin:/home/leoliu/anaconda3/bin:/home/leoliu/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

echo $LD_LIBRARY_PATH
/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/cusparse/lib:/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/nvjitlink/lib:/home/leoliu/anaconda3/lib::/home/leoliu/.mujoco/mujoco210/bin:/usr/lib/nvidia:/usr/lib/nvidia

echo $CUDA_HOME

nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0

which nvcc
/home/leoliu/anaconda3/envs/spatialvla/bin/nvcc

ninja                     1.11.1.3                 pypi_0    pypi
```
CUDA_HOME is empty.

githubLeoliu, 2025-02-24:
Based on above modification. I also did the following:

- update gcc version
- install cuda toolkit locally, instead of only in conda env
- Add cuda to PATH, LD_LIBRARY_PATH, CUDA_HOME

Now I have following setting:
```
(base) xxxx:~$ conda activate spatialvla
(spatialvla) xxxx:~$ echo $PATH
/home/leoliu/anaconda3/envs/spatialvla/bin:/home/leoliu/anaconda3/condabin:/home/leoliu/anaconda3/bin:/usr/local/cuda/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
(spatialvla) xxxx:~$ echo $LD_LIBRARY_PATH
/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/cusparse/lib:/home/leoliu/anaconda3/envs/spatialvla/lib/python3.10/site-packages/nvidia/nvjitlink/lib:/home/leoliu/anaconda3/lib:/usr/local/cuda/lib64:
(spatialvla) xxxx:~$ echo $CUDA_HOME
/usr/local/cuda/
(spatialvla) xxxx:~$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0
(spatialvla) xxxx:~$ which nvcc
/home/leoliu/anaconda3/envs/spatialvla/bin/nvcc
(spatialvla) xxxx:~$ gcc --version
gcc (Ubuntu 11.4.0-2ubuntu1~20.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
(spatialvla) xxxx:~$ conda list | grep ninja
ninja                     1.11.1.3                 pypi_0    pypi
(spatialvla) xxxx:~$ conda list | grep deepspeed
deepspeed                 0.15.3                   pypi_0    pypi
deepspeed-kernels         0.0.1.dev1698255861          pypi_0    pypi
(spatialvla) xxxx:~$ conda list | grep torch
torch                     2.5.1+cu121              pypi_0    pypi
torchaudio                2.5.1+cu121              pypi_0    pypi
torchvision               0.20.1+cu121             pypi_0    pypi
(spatialvla) xxxx:~$ conda list | grep transformers
transformers              4.47.0                   pypi_0    pypi
```

And I am still having the error.

Tavish9, 2025-02-24:
Hi, @githubLeoliu, after resetting the environment, remember to rebuild `fused_adam`.

run `rm -rf ~/.cache/torch_extensions/**/fused_adam` and relaunch the script.

Tavish9, 2025-02-24:
Okay, try reinstall deepspeed.

```bash
pip uninstall deepspeed
DS_BUILD_FUSED_ADAM=1 pip install deepspeed==0.15.3
ds_report
```
or 
```bash
pip uninstall deepspeed
git clone --branch v0.15.3 https://github.com/microsoft/DeepSpeed.git
cd DeepSpeed
DS_BUILD_FUSED_ADAM=1 pip3 install .
ds_report
```

Please share the error info, many thanks:)

githubLeoliu, 2025-02-24:
> Okay, try reinstall deepspeed.
> 
> pip uninstall deepspeed
> DS_BUILD_FUSED_ADAM=1 pip install deepspeed==0.15.3
> ds_report
> or
> 
> pip uninstall deepspeed
> git clone --branch v0.15.3 https://github.com/microsoft/DeepSpeed.git
> cd DeepSpeed
> DS_BUILD_FUSED_ADAM=1 pip3 install .
> ds_report
> Please share the error info, many thanks:)

The second approach works for me. Many thanks!