SpatialVLA/SpatialVLA #12: OOM during saving checkpoint

daerduoCarey, 2025-03-07:
Hi, dear authors,

I'm encountering GPU OOM during checkpoint saving. Have you encountered similar issue? Thanks!

batch-block7-00788:791989:835177 [6] include/alloc.h:103 NCCL WARN Cuda failure 2 'out of memory'
[rank6]: Traceback (most recent call last):
[rank6]:   File "/workspace/train/spatialvla_pretrain.py", line 702, in <module>
[rank6]:     main()
[rank6]:   File "/workspace/train/spatialvla_pretrain.py", line 670, in main
[rank6]:     train_result = trainer.train(resume_from_checkpoint=checkpoint)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/transformers/trainer.py", line 2164, in train
[rank6]:     return inner_training_loop(
[rank6]:   File "/usr/local/lib/python3.10/site-packages/transformers/trainer.py", line 2589, in _inner_training_loop
[rank6]:     self._maybe_log_save_evaluate(
[rank6]:   File "/usr/local/lib/python3.10/site-packages/transformers/trainer.py", line 3055, in _maybe_log_save_evaluate
[rank6]:     self._save_checkpoint(model, trial)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/transformers/trainer.py", line 3191, in _save_checkpoint
[rank6]:     self._save_optimizer_and_scheduler(output_dir)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/transformers/trainer.py", line 3301, in _save_optimizer_and_scheduler
[rank6]:     self.model_wrapped.save_checkpoint(output_dir)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/deepspeed/runtime/engine.py", line 3137, in save_checkpoint
[rank6]:     dist.barrier()
[rank6]:   File "/usr/local/lib/python3.10/site-packages/deepspeed/comm/comm.py", line 117, in log_wrapper
[rank6]:     return func(*args, **kwargs)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/deepspeed/comm/comm.py", line 408, in barrier
[rank6]:     return cdb.barrier(group=group, async_op=async_op)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/torch/_dynamo/eval_frame.py", line 632, in _fn
[rank6]:     return fn(*args, **kwargs)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/deepspeed/comm/torch.py", line 330, in barrier
[rank6]:     return torch.distributed.barrier(group=group, async_op=async_op, device_ids=device_ids)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/torch/distributed/c10d_logger.py", line 83, in wrapper
[rank6]:     return func(*args, **kwargs)
[rank6]:   File "/usr/local/lib/python3.10/site-packages/torch/distributed/distributed_c10d.py", line 4159, in barrier
[rank6]:     work = group.barrier(opts=opts)
[rank6]: torch.distributed.DistBackendError: NCCL error in: ../torch/csrc/distributed/c10d/NCCLUtils.hpp:317, unhandled cuda error (run with NCCL_DEBUG=INFO for details), NCCL version 2.21.5
[rank6]: ncclUnhandledCudaError: Call to CUDA function failed.
[rank6]: Last error:
[rank6]: Cuda failure 2 'out of memory'
W0307 03:11:14.748224 791918 site-packages/torch/distributed/elastic/multiprocessing/api.py:897] Sending process 791983 closing signal SIGTERM
W0307 03:11:14.749109 791918 site-packages/torch/distributed/elastic/multiprocessing/api.py:897] Sending process 791984 closing signal SIGTERM

DelinQu, 2025-03-08:
Hi! We haven't encountered this issue before. I suspect it might be related to the Deepspeed version. Could you please share the version of Deepspeed you're using?

daerduoCarey, 2025-03-08:
Yeah i'm also debugging hard into deepspeed. I'm using the same versions as in `requirements.txt`

HaomingSong, 2025-03-10:
Hi, this may be due to the memory pre-allocation in tensorflow, could you please try to add following code at the start of the `train/spatialvla_pretrain.py` or `train/spatialvla_pretrain.py`
```python
import tensorflow as tf
gpus = tf.config.experimental.list_physical_devices("GPU")
for gpu in gpus:
    try:
        tf.config.experimental.set_memory_growth(gpu, True)
    except:
        pass
```

daerduoCarey, 2025-03-10:
Thanks @HaomingSong . I will try. Somehow I find this error is stochastic. I mean, if I resume, the job can run and the checkpoints can be saved. I will report back my findings once I find the true reason. Thank you so much for all the helps!

closed 2025-03-11 by daerduoCarey

DelinQu, 2025-03-16:
I’ve simplified the code structure and resolved the dependency conflicts mentioned in issue [#19](https://github.com/SpatialVLA/SpatialVLA/issues/19). I started setting up the environment from scratch and trained the model, and everything is working fine. I believe the OOM issue during saving might be related to the buffer size. If you still encounter this problem, please feel free to let me know! Hope this helps ;)