SpatialVLA/SpatialVLA #14: Details about the released mix checkpoint

daerduoCarey, 2025-03-09:
Hi, dear authors,

for this model release (https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-mix-224-pt), what does this "mix" mean? Is this further finetuned on Fractal + Bridge given the spatialvla-4b-224-pt pretraining? Was this the model you evaluated and reported in Table I and II for SpatialVLA (fine-tuning)? Was the model finetuned on Fractal + Bridge together, or have you finetuned on each of the two separately? 

Thank you

DelinQu, 2025-03-09:
Hi! The mixed model was produced by fine-tuning the [SpatialVLA model](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-mix-224-pt/blob/main/IPEC-COMMUNITY/spatialvla-4b-224-pt) on the fractal and bridge dataset. This model is only used in our **TABLE V for Fine-tuning Ablations in Domain Datasets**. In Tables I and II, we fine-tune each **separately**. However, the experimental results show that there is little difference between fine-tuning separately and fine-tuning together.

Fine-tuning spatialvla-4b-224-pt on Bridge and Fractal is very brief, around 1-2 epochs, because the pretraining dataset already includes both of these datasets.

daerduoCarey, 2025-03-10:
Thank you! Will you release the two checkpoints for Table I and II?

DelinQu, 2025-03-11:
Yes! The checkpoint will be shared by Google Driver but not the official hugging face account. Please give me some time to find the corresponding checkpoints from the massive log files.

DelinQu, 2025-03-16:
Hi Prof,
We have checked all the logs and updated the model zoo. The current models are as follows:
<table>
  <tr>
    <th>Model Name</th>
    <th>Backbone</th>
    <th>VLA Model</th>
    <th>Note</th>
  </tr>
  <tr>
    <td>SpatialVLA-4B-224-pt</td>
    <td><a href="https://huggingface.co/google/paligemma2-3b-pt-224">google/paligemma2-3b-pt-224</a></td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-pt">spatialvla-4b-224-pt</a></td>
    <td>pretrained on openx and rh20t, TABLE I and II zero-shot, Fig.5 and 7</td>
  </tr>
  <tr>
    <td>SpatialVLA-4B-mix-224-pt</td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-pt">spatialvla-4b-224-pt</a></td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-mix-224-pt">spatialvla-4b-mix-224-pt</a></td>
    <td>fine-tuning on the fractal and bridge mixture dataset, Fig.5 and 7</a></td>
  </tr>
  <tr>
    <td>spatialvla-4b-224-sft-bridge</td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-pt">spatialvla-4b-224-pt</a></td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-bridge">spatialvla-4b-224-sft-bridge</a></td>
    <td>fine-tuning on the bridge dataset, testing on simple-env widowx-robot, TABLE I fine-tuning</a></td>
  </tr>
  <tr>
    <td>spatialvla-4b-224-sft-bridge</td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-pt">spatialvla-4b-224-pt</a></td>
    <td><a href="https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-fractal">spatialvla-4b-224-sft-fractal</a></td>
    <td>fine-tuning on the fractal dataset, testing on simple-env google-robot, TABLE II ine-tuning</a></td>
  </tr>
</table>

If you have any further questions or concerns, please let me know. I’ll do my best to assist!

closed 2025-03-16 by daerduoCarey