SpatialVLA/SpatialVLA #18: how about working with ur3 or ur5 robot arm

mmlingyu, 2025-03-13:
what i should do run with ours ur arm? uniform key ?

DelinQu, 2025-03-16:
Hi! If you want to run it with a new robot setup, you will need to recollect the data, convert it into the RLDS format, and fine-tune the mode. During the dataset loading, the normalization information will be automatically calculated. You can refer to this part for more details: [RLDS Dataset Loading](https://github.com/SpatialVLA/SpatialVLA/blob/0d034eaebb88acc9367d3300fdb7ba2d99dde96c/data/rlds.py#L205).

Keyustar, 2025-06-30:
> Hi! If you want to run it with a new robot setup, you will need to recollect the data, convert it into the RLDS format, 和 fine-tune the mode. During the dataset loading, the normalization information will be automatically calculated. You can refer to this part for more details: [get_dataset_statistics](https://github.com/SpatialVLA/SpatialVLA/blob/0d034eaebb88acc9367d3300fdb7ba2d99dde96c/data/rlds.py#L205)。你好！如果要使用新的机器人设置运行它，则需要重新收集数据，将其转换为 RLDS 格式，并微调模式。在数据集加载期间，将自动计算标准化信息。更多细节可以参考这部分： [get_dataset_statistics](https://github.com/SpatialVLA/SpatialVLA/blob/0d034eaebb88acc9367d3300fdb7ba2d99dde96c/data/rlds.py#L205)。

If there is a certain amount of dexterous hand data available, can the gripper be replaced with a dexterous hand to perform fine manipulation?