SpatialVLA/SpatialVLA #19: Hello, you have requested conflicting versions of PyTorch【 torch==2.5.1+cu121】 and TensorFlow【"tensorflow==2.15.0",】

Luo-Zhongwei, 2025-03-14:
in pytorch.org

# CUDA 11.8
conda install pytorch==2.5.1 torchvision==0.20.1 torchaudio==2.5.1  pytorch-cuda=11.8 -c pytorch -c nvidia
# CUDA 12.1
conda install pytorch==2.5.1 torchvision==0.20.1 torchaudio==2.5.1 pytorch-cuda=12.1 -c pytorch -c nvidia
# CUDA 12.4
conda install pytorch==2.5.1 torchvision==0.20.1 torchaudio==2.5.1 pytorch-cuda=12.4 -c pytorch -c nvidia

torch2.5.1 require CUDA 121.8,12.1,12.4 

in tensorflow.org

<html>
<body>

tensorflow-2.16.1 | 3.9-3.12 | Clang 17.0.6 | Bazel 6.5.0 | 8.9 | 12.3
-- | -- | -- | -- | -- | --
tensorflow-2.15.0 | 3.9-3.11 | Clang 16.0.0 | Bazel 6.1.0 | 8.9 | 12.2
tensorflow-2.14.0 | 3.9-3.11 | Clang 16.0.0 | Bazel 6.1.0 | 8.7 | 11.8
tensorflow-2.13.0 | 3.8-3.11 | Clang 16.0.0 | Bazel 5.3.0 | 8.6 | 11.8
tensorflow-2.12.0 | 3.8-3.11 | GCC 9.3.1 | Bazel 5.3.0 | 8.6 | 11.8

</body>
</html>

tf 2.15.0 require CUDA  12.2 ,SO it becames a problem

DelinQu, 2025-03-16:
Hi! The [dlimp](https://github.com/kvablack/dlimp) library has not been maintained for a long time, so the TensorFlow version might be out of date. A reliable solution is to comment out tensorflow==2.15.0 in the requirements file, install all other dependencies, and then install tensorflow==2.15.0 finally. Currently, using tensorflow==2.15.0 has not caused any problems. 

I'm try modifying the [dependency in dlimp](https://github.com/kvablack/dlimp/blob/5edaa4691567873d495633f2708982b42edf1972/setup.py#L8) to version 2.15.1 and testing the whole workflow to make sure compatibility.

closed 2025-03-16 by DelinQu

DelinQu, 2025-03-16:
Version 2.15.1 also works for dlimp. I have updated the dependencies in [dlimp](https://github.com/SpatialVLA/dlimp_custom).