SpatialVLA/SpatialVLA #21: Reproduce the results

MstarLioning, 2025-03-18:
Hello！Thank you very much for the inspiration your work has provided me! I would like to ask if you could kindly provide the code for evaluating RT-2-X on SimplerEnv, so as to reproduce the results mentioned in your paper?Looking forward to your reply!

DelinQu, 2025-03-18:
Hi [MstarLioning](https://github.com/MstarLioning), thanks for your attention. Please refer to [simplerenv-benchmark](https://github.com/SpatialVLA/SpatialVLA?tab=readme-ov-file#-simplerenv-benchmark) and [model-zoo](https://github.com/SpatialVLA/SpatialVLA?tab=readme-ov-file#-model-zoo) to reproduce the results in simpler-env benchmark.

DelinQu, 2025-03-18:
Oh! I see. Unfortunately, RT-2-X hasn’t released its open-source jax checkpoint, so our results were directly taken from OpenVLA.

MstarLioning, 2025-03-18:
@DelinQu Thank you very much for your reply! I understand your point, and I will try to initiate some other experiments. I wonder if I may trouble you with another question: I have noticed that in both SpatialVLA and OpenVLA, the LIBERO dataset is provided with the third-person perspective camera image labeled as “agentview_image.” However, LIBERO also offers a wrist-mounted camera “robot0_eye.” In your SpatialVLA work, was the decision to use “agentview_image” instead of “robot0_eye” made in order to control variables and enable comparisons with other models (such as OpenVLA)? If fine-tuning is necessary, would it be more advantageous to use the perspective from “robot0_eye”?

Lastly, I would like to ask one more thing: have you considered defining new scenarios and tasks beyond the LIBERO suite to evaluate SpatialVLA? I observed some discussions between you and others in the OpenVLA issues section, and I am curious if you had attempted to construct and evaluate new LIBERO scenarios outside of the four officially provided suites. After our attempts to set up new scenarios, the evaluation results on any Fine-tuning-OpenVLA were very poor, and we hope to evaluate on SpatialVLA as well. Do you think this work is meaningful and necessary?

I eagerly await your reply and sincerely appreciate your help!

DelinQu, 2025-03-18:
> using agentview_image” instead of “robot0_eye”

Thank you for your insightful question! To ensure a fair comparison with OpenVLA on LIBERO, we only used the “agentview_image.” However, in our recent experiments, we found that incorporating the wrist camera (“robot0_eye”) significantly improves the model’s performance (increasing to around 92% on average). Using **multi-view inputs can partially alleviate the model’s inability to perceive depth, leading to more accurate performance**. We plan to release the enhanced LIBERO testing code with multi-view and state input support soon.

> new scenarios and tasks beyond the LIBERO suite? 

In our **simulation experiments** (Fig. 4: Experiment Setup), we only evaluated the zero-shot and transfer capabilities of spatialvla on SimpleEnv and LIBERO benchmarks. Whether SpatialVLA can perform well in entirely new scenarios, I believe that fine-tuning in those scenarios is essential to achieve satisfactory results—this is a common limitation shared by most VLAs. Therefore, when setting up new scenarios and evaluating SpatialVLA, I **expect SpatialVLA to perform similarly to OpenVLA, which means the results might still be relatively poor**.

Moreover, I think the LIBERO benchmark does not fully measure a model’s generalizability but rather its tendency to overfit to specific tasks. More recent benchmarks like VLA-Bench provide a more comprehensive evaluation.

MstarLioning, 2025-03-18:
Thank you for your comprehensive reply; it really helped me a lot! I sincerely wish you all the best and continued success.

closed 2025-03-18 by MstarLioning