SpatialVLA/SpatialVLA #22: GUI problem:: [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE

Luo-Zhongwei, 2025-03-19:
hi  I met two problem：
1）[2025-03-17 14:46:18.057] [svulkan2] [error] Failed to denoise OptiX Error: OPTIX_ERROR_INVALID_VALUE
I think I already install  optix

-----

2)[2025-03-15 05:52:22.909] [svulkan2] [error] GLFW error: X11: Failed to open display 
[2025-03-15 05:52:22.909] [svulkan2] [warning] Continue without GLFW.

but I think I already set this:
-----
Xvfb :1 -screen 0 1024x768x24 +extension GLX +render -noreset &
export DISPLAY=:1
nohup bash scripts/run_spatialvla.sh > /root/autodl-tmp/lzw/SimplerEnv-OpenVLA/test_vla.txt 2>&1 &

DelinQu, 2025-03-20:
Hi, please refer to https://github.com/simpler-env/SimplerEnv/issues/64. I encountered this problem and validated the solution in a local machine.

Luo-Zhongwei, 2025-03-21:
let me try it