SpatialVLA/SpatialVLA #24: Inquiry Regarding the Hyperparameter “unnorm_key”

MstarLioning, 2025-03-20:
Hello, sorry to bother you again. As a beginner in VLA, I’m curious about the hyperparameter unnorm_key in SpatialVLA and its role (mapping it back to real space using the formula 0.5 * (normalized_actions + 1) * (action_high - action_low) + action_low).

My main confusion is why a single set of model parameters can have different unnorm_key values. In my understanding, besides mimicking trajectories, the VLA training process also requires the model to "understand" the ratio between its output values and the actual spatial displacement/rotation. If that’s the case, shouldn’t a unified set of parameters only learn (or comprehend) a single scaling factor between the model’s output and the real space?

What confuses me even more is why the unnorm_key is explicitly specified. After examining the processor_config.json file of SpatialVLA, I noticed that the unnorm_key refers to a series of statistical features, and these statistics are obtained during training. So why isn’t the mapping from normalized data to real-space translation/rotation a fixed relationship? Instead, it seems to vary with the dataset. In other words, what should be done in an out-of-distribution situation when a new sample arrives for prediction and cannot be classified into a particular dataset?

I have observed that in other VLAs (π0, oepnvla, DexGraspVLA), the unnormalization is generally represented in this manner. Why not implement it as a fixed mapping? Could you please share your thoughts or analysis on this hyperparameter?

I look forward to your reply and hope to receive your guidance!

DelinQu, 2025-03-20:
您好！
主要原因是openx不同数据集的机械臂构型很不一样，而且尺度差别非常大，如果都用统一的mean std会导致某些数据集分布在非常小或非常大的范围内，具身的数据相对于imagenet这种cv的image差别还是太大了。在训练的时候数据送到一个dataloader分发给模型训练，数据根据不同mean std和normalized 到[-1,1]的范围。推理的时候unnomalize出来。当然如果不做pretrain，只在某些小数据集上tuning这个问题会小很多。

DelinQu, 2025-03-20:
1. 一般来说unnorm_key是针对某一个数据集的，而不是机械臂，最好是每个数据集统计一下。
2. rlds在训练前会统计mean std，对应代码是https://github.com/SpatialVLA/SpatialVLA/blob/53a59f91ad90e5b8d4eb409e9ebae311e847dd98/data/utils/data_utils.py#L244
3. fine-tuning的时候，一般是固定几个小数据集，一个数据集训一次模型，那么这个key直接存到模型里就好了，不需要维护一个很大的dict。

DelinQu, 2025-03-20:
如果没有数据集tuning的话，可以试一试

MstarLioning, 2025-03-20:
收到，谢谢您！我会尽快去尝试！再次感谢您耐心的解答！

closed 2025-03-20 by MstarLioning