SpatialVLA/SpatialVLA #25: hi, bash scripts/run_spatialvla.sh get such results on H20,is that basiclly correct? and why apple_in_drawer is empty

Luo-Zhongwei, 2025-03-21:
[image]

DelinQu, 2025-03-24:
Yes, and the result of sft model should be consistent with the result files provided in the hugging face repo: [spatialvla-4b-224-sft-bridge](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-bridge/blob/main/simplerenv.md) and [spatialvla-4b-224-sft-fractal](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-fractal/blob/main/simplerenv.md).

Similar to most papers, e.g., HPT and TraceVLA, we omitted the Open Top Drawer and Place Apple from our evaluation, since the vast majority of policies achieved scores approaching 0 on this task. Integrating with state inputs would help to fix the issue.

Luo-Zhongwei, 2025-03-25:
> 是的，并且 sft 模型的结果应该与 hugging face repo 中提供的结果文件一致：[spatialvla-4b-224-sft-bridge](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-bridge/blob/main/simplerenv.md)和[spatialvla-4b-224-sft-fractal](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-sft-fractal/blob/main/simplerenv.md)。
> 
> 与大多数论文（例如 HPT 和 TraceVLA）类似，我们从评估中排除了 Open Top Drawer 和 Place Apple，因为绝大多数策略在此任务上的得分接近 0。与状态输入相结合将有助于解决此问题。

what is more,could plz give more hints about " Integrating with state inputs would help to fix the issue."