SpatialVLA/SpatialVLA #26: How long it takes to fine-tune Google Fractual and Bridge V2 Dataset?

LukeLIN-web, 2025-03-24:
TABLE V:Fine-tuning Ablations in Domain Datasets. Pretrained models are full parameter fine-tuned in individual Google Fractual and Bridge V2 Dataset.

I wonder how many epochs we need to finetune on Google Fractual and Bridge V2 Dataset?  I checked paper but seems not mention it. 

Did you think LoRA-tuning is possible for Google Fractual and Bridge V2 Dataset?

DelinQu, 2025-03-26:
Hi, it takes about 1~3 epochs in both Google Fractual and Bridge V2 Dataset, with lr = 5e-6. We use full-prams tuning if the dataset is large-scale. I think LORA-tuning with extra parameters is unnecessary because the pertaining dataset of [spatialvla-4b-224-pt](https://huggingface.co/IPEC-COMMUNITY/spatialvla-4b-224-pt) has both of these datasets already. The initial ACC is very high (about 95%) if you start a full-params tuning.

LukeLIN-web, 2025-04-10:
> Hi, it takes about 1~3 epochs in both Google Fractual and Bridge V2 Dataset, with lr = 5e-6. We use full-prams tuning if the dataset is large-scale. I think LORA-tuning with extra parameters is [...]

bridge has 60k trajctary, Dataset Size = 2135488 . I wonder whether one epoch is 2135488 pair of image/instruction?

LukeLIN-web, 2025-05-04:
> Hi, it takes about 1~3 epochs in both Google Fractual and Bridge V2 Dataset, with lr = 5e-6. We use full-prams tuning if the dataset is large-scale. I think LORA-tuning with extra parameters is [...]

Thank you, I wonder how many GPUs and batch size you are using?

closed 2025-05-28 by DelinQu