SpatialVLA/SpatialVLA #27: the dataset about LIBERO experiment

szl2001, 2025-03-25:
I noticed in other issues that you conducted LIBERO tests following the approach of OpenVLA. In OpenVLA, the researchers used a reduced version of the LIBERO dataset. I would like to know whether SpatialVLA uses the original LIBERO dataset or the reduced version from OpenVLA. Thank you.

DelinQu, 2025-03-26:
Thanks for your attention! In our initial experiments, we only performed basic filtering according to our previous real robot deployment experience (removing zero actions) but didn't implement OpenVLA's full modifications to the LIBERO dataset. After revisiting recently, we've conducted additional fine-tuning using the OpenVLA dataset version, which yielded performance improvements. Hence, we recommend applying OpenVLA's data filtering approach to enhance model performance. We'll release the fine-tuning configurations and evaluation codes for LIBERO experiments in our upcoming update.

szl2001, 2025-03-26:
Thank you for your answer. I am looking forward to your new experiments and results

szl2001, 2025-03-26:
> Thanks for your attention! In our initial experiments, we only performed basic filtering according to our previous real robot deployment experience (removing zero actions) but didn't implement [...]

Thank you for your answer. I am looking forward to your new experiments and results