SpatialVLA/SpatialVLA #32: Module Import and Video Writing Issues

MstarLioning, 2025-03-27:
Hello, sorry to trouble you! I used your OpenVLA-SimplerEnv previously without any issues, but when I referenced SpatialVLA into SimplerEnv, I encountered problems with video recording and some strange output messages. The specific information is: "The error log indicates that there is a module import error where the Octo module fails to import due to a missing 'octo.model' module. Additionally, a BrokenPipeError occurs when using mediapy to write a video, followed by a RuntimeError during video file creation with ffmpeg. This RuntimeError is related to missing version information for libncursesw and a symbol lookup error in libgobject-2.0.so.0." The relevant pip list, the complete output, and the highlighted problematic output have been attached after this issue. I look forward to your reply!

Octo problem:
'''
Some kwargs in processor config are unused and will not have any effect: action_config, statistics, num_obs_steps, bin_policy, intrinsic_config, action_chunk_size, obs_delta. 
Octo is not correctly imported.
No module named 'octo.model'
'''

Broken problem:
'''
59 {'elapsed_steps': 60, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': False, 'consecutive_grasp': False, 'src_on_target': True, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', True)]), 'success': True}
Traceback (most recent call last):
  File "/home/ch/.conda/envs/spatialvla/lib/python3.10/site-packages/mediapy/__init__.py", line 1749, in write_video
    writer.add_image(image)
  File "/home/ch/.conda/envs/spatialvla/lib/python3.10/site-packages/mediapy/__init__.py", line 1653, in add_image
    if stdin.write(data) != len(data):
BrokenPipeError: [Errno 32] Broken pipe

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lsf_storage/homes/ch/claude/SpatialVLA/SimplerEnv-OpenVLA/simpler_env/main_inference.py", line 85, in <module>
    success_arr = maniskill2_evaluator(model, args)
  File "/home/lsf_storage/homes/ch/claude/SpatialVLA/SimplerEnv-OpenVLA/simpler_env/evaluation/maniskill2_evaluator.py", line 227, in maniskill2_evaluator
    run_maniskill2_eval_single_episode(
  File "/home/lsf_storage/homes/ch/claude/SpatialVLA/SimplerEnv-OpenVLA/simpler_env/evaluation/maniskill2_evaluator.py", line 173, in run_maniskill2_eval_single_episode
    write_video(video_path, images, fps=5)
  File "/home/lsf_storage/homes/ch/claude/SpatialVLA/SimplerEnv-OpenVLA/simpler_env/utils/visualization.py", line 28, in write_video
    media.write_video(path, images_npy, fps=fps)
  File "/home/ch/.conda/envs/spatialvla/lib/python3.10/site-packages/mediapy/__init__.py", line 1747, in write_video
    with VideoWriter(path, shape=shape, dtype=dtype, **kwargs) as writer:
  File "/home/ch/.conda/envs/spatialvla/lib/python3.10/site-packages/mediapy/__init__.py", line 1614, in __exit__
    self.close()
  File "/home/ch/.conda/envs/spatialvla/lib/python3.10/site-packages/mediapy/__init__.py", line 1671, in close
    raise RuntimeError(f"Error writing '{self.path}': {s}")
RuntimeError: Error writing 'results/spatialvla-4b-224-pt-0.8/bridge_table_1_v1/arm_pd_ee_target_delta_pose_align2_gripper_pd_joint_pos/PutCarrotOnPlateInScene-v0/rob_0.147_0.028_rot_0.000_-0.000_3.142_rgb_overlay_bridge_real_eval_1/success_obj_episode_0_moved_correct_obj_True_moved_wrong_obj_False_is_src_obj_grasped_True_consecutive_grasp_True_src_on_target_True.mp4': /usr/bin/ffmpeg: /home/lsf_storage/software/anaconda3/lib/libncursesw.so.6: no version information available (required by /lib/x86_64-linux-gnu/libcaca.so.0)
/usr/bin/ffmpeg: /home/lsf_storage/software/anaconda3/lib/libncursesw.so.6: no version information available (required by /lib/x86_64-linux-gnu/libcaca.so.0)
/usr/bin/ffmpeg: symbol lookup error: /lib/x86_64-linux-gnu/libgobject-2.0.so.0: undefined symbol: ffi_type_uint32, version LIBFFI_BASE_7.0

2025-03-27 13:03:30.756664: E external/local_xla/xla/stream_executor/cuda/cuda_dnn.cc:9261] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered

'''

pip list:
'''
Package                      Version               Editable project location
---------------------------- --------------------- -----------------------------------------------------------------------------------
absl-py                      2.2.1
accelerate                   1.0.1
aiohappyeyeballs             2.6.1
aiohttp                      3.11.14
aiosignal                    1.3.2
annotated-types              0.7.0
apache-beam                  2.54.0
array_record                 0.7.1
asttokens                    3.0.0
astunparse                   1.6.3
async-timeout                5.0.1
attrs                        25.3.0
beautifulsoup4               4.13.3
cachetools                   5.5.2
certifi                      2025.1.31
chardet                      4.0.0
charset-normalizer           3.4.1
chex                         0.1.89
click                        8.1.8
cloudpickle                  2.2.1
cmake                        3.27.6
coacd                        1.0.6
[... 388 lines cut ...]
22 {'elapsed_steps': 23, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
23 {'elapsed_steps': 24, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
24 {'elapsed_steps': 25, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
25 {'elapsed_steps': 26, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
26 {'elapsed_steps': 27, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
27 {'elapsed_steps': 28, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
28 {'elapsed_steps': 29, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
29 {'elapsed_steps': 30, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
30 {'elapsed_steps': 31, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': True, 'consecutive_grasp': True, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
[... cut ...] , True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
44 {'elapsed_steps': 45, 'moved_correct_obj': True, 'moved_wrong_obj': False, 'is_src_obj_grasped': False, 'consecutive_grasp': False, 'src_on_target': False, 'episode_stats': OrderedDict([('moved_correct_obj', True), ('moved_wrong_obj', False), ('is_src_obj_grasped', True), ('consecutive_grasp', True), ('src_on_target', False)]), 'success': False}
45 
'''

MstarLioning, 2025-03-27:
The deepest path in the results folder is at:

/home/lsf_storage/homes/ch/claude/SpatialVLA/SimplerEnv-OpenVLA/results/spatialvla-4b-224-pt-0.8/frl_apartment_stage_simple/arm_pd_ee_delta_pose_align_interpolate_by_planner_gripper_pd_joint_target_delta_pos_interpolate_by_planner/OpenBottomDrawerCustomInScene-v0_shader_dir_rt/rob_0.65_-0.2_rot_0.000_-0.000_3.142_rgb_overlay_None

However, there are no videos inside.

DelinQu, 2025-03-27:
You could try reinstalling ffmpeg with apt or conda. I think the issue is caused by the lacking of ffmpeg

MstarLioning, 2025-03-27:
Thank you! The issue has been completely resolved.

closed 2025-03-27 by MstarLioning