SpatialVLA/SpatialVLA #38: 关于 SpatialVLA 生成 Token 数量的问题请教

MstarLioning, 2025-04-23:
尊敬的作者您好：
        我在采用您Readme主页的代码时通过tokens = processor.tokenizer.convert_ids_to_tokens(gen_ids.tolist())生成了输出的token，但是观察输出token一共有13个维度，分别是：`['<ACTION01443>', '<ACTION05772>', '<ACTION08193>', '<ACTION01699>', '<ACTION06044>', '<ACTION08193>', '<ACTION01973>', '<ACTION06077>','<ACTION08193>', '<ACTION01973>', '<ACTION05550>', '<ACTION08193>','<eos>']`请问这正常么？按说不应该只输出∆T,∆R,G的7维Token么？

我们使用的代码全文是：
```
import torch
from PIL import Image
from transformers import AutoModel, AutoProcessor

model_name_or_path="IPEC-COMMUNITY/spatialvla-4b-224-pt"
processor = AutoProcessor.from_pretrained(model_name_or_path, trust_remote_code=True)
model = AutoModel.from_pretrained(model_name_or_path, trust_remote_code=True, torch_dtype=torch.bfloat16).eval().cuda()

inputs = processor(images=[Image.open("coke.png")], text="pick coke can", return_tensors="pt").to("cuda")

outputs = model.generate(
    **inputs,
    max_new_tokens=20,
    output_attentions=True,
    return_dict_in_generate=True,
    num_beams=1
)

input_len = inputs["input_ids"].shape[-1]
gen_ids = outputs.sequences[0][input_len:]
tokens = processor.tokenizer.convert_ids_to_tokens(gen_ids.tolist())
```

提前感谢您百忙之中抽空帮助我解答疑惑，万分期待能得到您的指导！

DelinQu, 2025-04-23:
[image]

这里的action chunk是4，加上最后eos token是 3 * 4 + 1 = 13

MstarLioning, 2025-04-23:
收到收到！理解了！谢谢您！