SpatialVLA/SpatialVLA #49: Gradient Calculation Validation: Are Our Results Reasonable?

MstarLioning, 2025-06-02:
Hello!We're analyzing SpatialVLA gradients using Fractal dataset samples and want to confirm our implementation follows the correct methodology.

## Our Implementation

### **Computation Target**: Single Fractal episode step
- **Input**: RGB image (224×224) + natural language instruction
- **Ground Truth**: 7D action vector `[world_vector(3), rotation_delta(3), gripper_closedness(1)]`
- **Sample**: `"pick rxbar chocolate from bottom drawer and place on counter"`

### **Computation Strategy**: 
1. **Action Tokenization**: Used `suffix_actions` parameter in processor
   ```python
   processed_inputs = model_wrapper.processor(
       images=[image], 
       text=[instruction], 
       suffix_actions=expert_action_7d.reshape(1, -1),  # 7D action
       return_tensors="pt"
   )
   ```

2. **Loss Calculation**: Auto CrossEntropyLoss from model forward pass
   ```python
   outputs = model(**processed_inputs)  # labels auto-generated
   loss = outputs.loss  # CrossEntropyLoss on action tokens only
   ```

3. **Gradient Computation**: Standard PyTorch backward
   ```python
   loss.backward()
   ```

## Results Obtained

| Metric | Value | 
|--------|-------|
| **Loss** | 13.755 |
| **Parameters with gradients** | 735 |
| **Average gradient norm** | 25.78 |
| **Total gradient norm** | 18,949 |

### Key Observations:
- Only action token positions contribute to loss (non-action tokens masked with -100)
- Vision encoder gradients: ~0.1-10 range
- Language model gradients: ~1-100 range  
- Position embedding gradients: ~2-76 range

## Questions

1. **Are these gradient magnitudes reasonable** for single-sample training?
2. **Is our action tokenization approach correct** using `suffix_actions`?
3. **Should we expect loss ~13.8** for a challenging manipulation task?
4. **Are we missing any normalization** or scaling factors?

## Environment
- Model: `IPEC-COMMUNITY/spatialvla-4b-224-pt`
- Data: Fractal dataset (Google Robot)
- Precision: Mixed precision (bfloat16)

Any guidance on expected gradient ranges or methodology corrections would be greatly appreciated!

DelinQu, 2025-06-07:
Hi,
I'm so sorry for my late reply. Here are the loss/accuracy metrics when fine-tuning SpatialVLA on the Fractal dataset. The model maintains high accuracy and low L1 error.
[image]

However, potential problems might lie in: `apply_rt1_dataset_transform`, `Normalization` and `data augmentation`. In [processing_spatialvla.py#L216](https://github.com/SpatialVLA/SpatialVLA/blob/f8904f5962e9bf0fe733e7abadd65b02ae1658b4/model/processing_spatialvla.py#L219C9-L219C19), we use unnorm_key to denormalize model outputs to match GT action scale. However, when computing token IDs for GT actions at [processing_spatialvla.py#L108](https://github.com/SpatialVLA/SpatialVLA/blob/f8904f5962e9bf0fe733e7abadd65b02ae1658b4/model/processing_spatialvla.py#L108), the code assumes actions are already normalized. Could you clarify your GT action normalization process?

MstarLioning, 2025-06-09:
您好！非常感谢您的回复和建议！我们根据您提到的几个问题分别check和修正了一下，目前已经实现Loss与Accuracy与您提供图片一样的数量级了，但是虽然数量级接近，具体的数值在10000个样本上的计算仍略高于您展示的图表。故而想请教您一下您使用的模型是IPEC-COMMUNITY/spatialvla-4b-224-pt还是IPEC-COMMUNITY/spatialvla-4b-224-sft-fractal呢？或者不知道我们有没有什么遗漏的细节？期待您的进一步指导！

[image]

DelinQu, 2025-06-18:
这个loss是从COMMUNITY/spatialvla-4b-224-pt微调到IPEC-COMMUNITY/spatialvla-4b-224-sft-fractal，初期loss略有波动几十个epoch后会稳定在高ACC。我这边check一下上传的权重，确实是sft-fractal微调的结果。我还是倾向于说数据预处理，图像增强等等可能没完全对齐。

closed 2025-06-18 by MstarLioning

Qinruojun, 2026-11-13:
I met the same problem too, and I found that if the gt action is encoded and then decoded , the decoded action can not replay successfully