SpatialVLA/SpatialVLA #55: Dataset statistics mismatched

Boltzmachine, 2025-06-24:
Hi I downloaded the OXE dataset. Since you do not have very detailed instructions of dataset preparation, I mostly follow https://github.com/openvla/openvla?tab=readme-ov-file#vla-pretraining-datasets and [this script](https://github.com/moojink/rlds_dataset_mod/blob/main/prepare_open_x.sh).

However, I found some datasets mismatch the statistics reported in the paper, regarding the number of samples. which are listed below

|dataset| #trajectory of mine | #trajectory in the paper|
|---|---|---|
|roboturk| 1796| 1995 |
|taco play| 3242 | 3603 |
|berkerley autolab| 896 | 1000 |
|toto| 902 | 1003 |
|viola| 135 | 150 |
|jaco play| 976 | 1085 |
|berkerley table routing| 1482 | 1647|

Other datasets are fine.

Do you have any idea of what happens? 🙏

DelinQu, 2025-06-28:
Thanks for reaching out and for your careful comparison of the dataset statistics. After looking into this, I believe the difference isn't due to data filtering but rather stems from the default train/test split (0.9 ratio) in the RLDS dataset: https://github.com/SpatialVLA/SpatialVLA/blob/18fd74b2a633ec8d9ec7aadcd803969555cc9fbd/data/rlds.py#L233C5-L233C12. This would explain why your numbers are consistently about 90% of the reported figures across different datasets.

While the counts aren't identical, this small variance in dataset size shouldn't meaningfully impact model performance. The key characteristics and quality of the data remain consistent.

Boltzmachine, 2025-06-30:
So does that mean you use both train/test splits for the training of SpatialVLA?

DelinQu, 2025-06-30:
Yes, SpatialVLA uses weighted sampling across all available datasets without a held-out test split.

Boltzmachine, 2025-07-01:
Ah I see. I found it is actually that I use `len(dataset)` somewhere for dataset statistics, and test split is not included for some datasets. Really appreciate your helps!

closed 2025-07-01 by Boltzmachine