SpatialVLA/SpatialVLA #56: Question about Ego3D position encoding

geyan21, 2025-06-26:
Hi, great work! I wonder how you handle multi-view egocentric point clouds. During training, do you load all views each step and simply concatenate them together, or load one view each time to avoid merging? I'm asking this because multiple egocentric point clouds may cause confusion if the same object is present in multiple views but with different coordinate systems. Or the model can just learn to register different views during training.

DelinQu, 2025-06-28:
Hi [geyan21](https://github.com/geyan21)! 
Currently, following OpenVLA's approach, SpatialVLA uses OpenX for pretraining with only a single view.  In many multi-view scenarios, 3D information becomes less critical as different camera perspectives can complement each other effectively. Processing full 3D point clouds from multiple views would significantly increase computational overhead.

geyan21, 2025-07-02:
Hi @DelinQu, thank you for the clarification! Referring `"SpatialVLA uses OpenX for pretraining with only a single view."`, do you just randomly sample a single view from multiple views in data loading each time? What would happen if the selected view has very poor depth quality, or do you just perform depth estimation for all images even though they have real-world depth to avoid this issue?

DelinQu, 2025-07-08:
Image views are configurable in https://github.com/SpatialVLA/SpatialVLA/blob/18fd74b2a633ec8d9ec7aadcd803969555cc9fbd/data/oxe/configs.py#L56 rather than being randomly selected. We do inference depth for all the image observations even though the depth exists already, because aligning the depth and depth to RGB image pixels can be tricky.

geyan21, 2025-07-08:
Thank you for the reply! Just curious if to leverage multi-view information, would it be a good idea to encode multiple egocentric-view point clouds directly?